olikraus/U8g2_for_Adafruit_GFX #5: newline handling and Adafruit_GFX cursor update missing?

ZinggJM, 2018-03-21:
Oliver, thank you for this welcome library!

I am still starting my tests, but it looks like:
println() does not start a new line
getCursorX() and getCursorY() of the referenced display class do not report the actual cursor.

thank you

Jean-Marc

olikraus, 2018-03-21:
> println() does not start a new line

Yes, because I do not know what should be done here. Maybe you should specify this more clearer in terms of what should happen to tx/ty.

Actually I did not implement any newline behaviour in u8g2 because it does not make so much sense on a graphical interface. So my question: What shell happen when there is a newline? How do you define "newline"?

> getCursorX() and getCursorY() of the referenced display class do not report the actual cursor.
(not needed, as tx and ty are public).

hmm.. thats sounds like a do it or leave it... The question is more, you want it or not? ;-)

ZinggJM, 2018-03-21:
newline should set tx to zero and increase ty with height of glyph + some spacing.
more exactly \r should set tx to zero, \n should set tx to zero and increase ty.
I can't be more specific, would need to look what Adafruit_GFX does.

On second thought, U8g2_for_Adafruit_GFX could provide getCursorX() and getCursorY() read accessors to tx and ty, would look familiar to Adafruit_GFX users.

Thank you.

ZinggJM, 2018-03-25:
Thank you for the additions.

What about newline handling? This is what Adafruit_GFX does:

#if ARDUINO >= 100
size_t Adafruit_GFX::write(uint8_t c) {
#else
void Adafruit_GFX::write(uint8_t c) {
#endif
    if(!gfxFont) { // 'Classic' built-in font

        if(c == '\n') {                        // Newline?
            cursor_x  = 0;                     // Reset x to zero,
            cursor_y += textsize * 8;          // advance y one line
        } else if(c != '\r') {                 // Ignore carriage returns
            if(wrap && ((cursor_x + textsize * 6) > _width)) { // Off right?
                cursor_x  = 0;                 // Reset x to zero,
                cursor_y += textsize * 8;      // advance y one line
            }
            drawChar(cursor_x, cursor_y, c, textcolor, textbgcolor, textsize);
            cursor_x += textsize * 6;          // Advance x one char
        }

    } else { // Custom font

        if(c == '\n') {
            cursor_x  = 0;
            cursor_y += (int16_t)textsize *
                        (uint8_t)pgm_read_byte(&gfxFont->yAdvance);
        } else if(c != '\r') {
            uint8_t first = pgm_read_byte(&gfxFont->first);

olikraus, 2018-03-25:
yes, something like this. But I did not intend to support wrapping.

FrenchLab47, 2018-04-05:
Hello Oliver,

I am testing U8g2_for_Adafruit_GFX on an ESP32 with an RGB matrix, and it works pretty well. I am delighted to finally be able to display French accents on dynamic and scrollable texts. Congratulations for your work.

I have just a question:
With the originals fonts of Adafruit GFX, I can display all basic colors. But if I select a U8g2 font, it can not be displayed in blue, or in colors that contain blue. I guess there must be a little oversight somewhere in the code. Do you have an idea ?
Thank you very much.

I also asked the same thing on the Arduino forum :)

olikraus, 2018-04-05:
I saw the forum post, but we can discuss here also. I had started with a 8 bit color value, but switched to 16 bit color value now. Maybe the stable release is not aware about the change. 

Hm... it shouled be fixed in release 1.1.0: https://github.com/olikraus/U8g2_for_Adafruit_GFX/issues/1

olikraus, 2018-04-05:
ok, fixed and released...

closed 2018-04-05 by olikraus