olikraus/U8g2_for_Adafruit_GFX #6: typo in write(uint8_t v) method

av1024, 2018-04-04:
Hi!
Does not compiled for me.
Looks like missed 'u8g2.' in write function. 

```size_t write(uint8_t v) {
      uint16_t e = utf8_next(v);
      int16_t delta;
      
      if ( e == '\n' )
      {
	tx = 0;
// FIXED HERE:
        ty += u8g2.font_info.ascent_para - u8g2.font_info.descent_para;
      }
```
PS: I try to adapt this lib to RTL8710 with custom eInk library, so I use QtCreator/QBS for build. May be this is non-arduino related issue?

olikraus, 2018-04-04:
You mean this line?
https://github.com/olikraus/U8g2_for_Adafruit_GFX/blob/master/src/U8g2_for_Adafruit_GFX.h#L185

Did you use the version from Arduino IDE?

Or did you just download zip from here?

av1024, 2018-04-04:
> You mean this line?
Yes

> Did you use the version from Arduino IDE?
No. I use `git clone` (a hour before creating issue) so it latest version

av1024, 2018-04-04:
Can confirm that code compiled with fixed typo work with custom drawing engine (for now - Paint class from Waveshare examples) with 2.13" eInk display.

Complied via QtCreator's "makefile" - .qbs script using forked rtlDuino library.

BTW: As a thoughts out loud. Looks like Adafruit's GFX is not main part of lib. May be a simple "draw_H(V)_Line" pointers would be enough?

olikraus, 2018-04-04:
> May be a simple "draw_H(V)_Line" pointers would be enough?

You mean?

olikraus, 2018-04-05:
fixed, tested and released

closed 2018-04-05 by olikraus

av1024, 2018-04-06:
>> May be a simple "draw_H(V)_Line" pointers would be enough?
>You mean?

For now only two functions from GFX used. So it might be an option to remove dependency of Adafruit_GFX and store two pointers for drawing functions.

olikraus, 2018-04-06:
True, but the lib was created with Adafruit in mind only.
In this case it should be a new github library.