olikraus/U8g2_for_Adafruit_GFX #8: Transparency not working [Misleading examples?]

Tuet, 2018-07-31:
Hi, I tried to put a font on a non-black background and want it to be transparent. So I did first setFontMode(1), then selected a Font and then printed a string - according to your examples:

```
  u8g2_for_adafruit_gfx.setFontMode(1);                 // use u8g2 transparent mode (this is default)
  u8g2_for_adafruit_gfx.setFontDirection(0);            // left to right (this is default)
  u8g2_for_adafruit_gfx.setForegroundColor(WHITE);      // apply Adafruit GFX color
  u8g2_for_adafruit_gfx.setFont(u8g2_font_helvR14_tf);  // select u8g2 font
```
I didn't get the expected results all the time - some portions where still black instead of transparent. I looked into your library code and found that the mistake is not in the library itself, but in the understanding of the inner workings: The FontMode seems to be a setting stored in the data of to the selected font - and not globally in the lib as I expected. So the order of calling the functions is important: You have to select the Font first, and then set the desired FontMode - with this knowledge everything is working fine and the Font renders with transparent background. However, the examples you provide are misleading, as you select the Font last and not first. So (if I am right with my assumption) can you change the example codes to do the selection of the font first?

olikraus, 2018-07-31:
ok... makes sense.

hznupeter, 2019-08-07:
this problem sitll existed

olikraus, 2019-08-07:
Which problem?

hznupeter, 2019-08-07:
you should set the font before set the fontmode.

olikraus, 2019-08-08:
Hmmm No, you should set the font mode after setting the font. But as discussed above, it is not a bug