olikraus/U8g2_for_Adafruit_GFX #10: Support of Adafruit NeoMatrix library

lobl, 2018-10-13:
Can this library be used to bring U8g2 fonts on displays built with led stripes which are driven by the Adafruit NeoMatrix library ([https://github.com/adafruit/Adafruit_NeoMatrix](https://github.com/adafruit/Adafruit_NeoMatrix))?

olikraus, 2018-10-13:
Could be, if this lib is compatible to the Adafruit_GFX lib. May you should just try.
Remember to replace the include statements.

Galaxy-Man, 2022-07-11:
I know it's 4 years late and not related to this library. 
Answer: However it does work with NeoMatrix.
matrix.fillScreen(0);
u8g2_for_adafruit_gfx.setFont(u8g2_font_5x7_mf);
u8g2_for_adafruit_gfx.drawStr(x, y, "1234567890");
u8g2_for_adafruit_gfx.setForegroundColor(RED);
matrix.show();