olikraus/U8g2_for_Adafruit_GFX #12: Enable large font support

olikraus, 2018-10-29:
The support for large fonts was disabled. This bugfix should enable large fonts.
Background: Request was to support Chinese fonts. However, large Chinese fonts are not found, due to this bug.

closed 2018-10-29 by olikraus

Veter1972, 2018-10-31:
Hello!
I instaled a new release of U8g2 library for Adafruit GFX, release 1.5.1.
Unfortunately, after all new libraries installed I still get some errors during compiling:
C:\Users\Lenovo\AppData\Local\Temp\ccKTXxRq.ltrans1.ltrans.o: In function `main':

ccKTXxRq.ltrans1.o:(.text.startup+0x114): undefined reference to `u8g2_font_wqy16_t_gb2312b'

ccKTXxRq.ltrans1.o:(.text.startup+0x116): undefined reference to `u8g2_font_wqy16_t_gb2312b'

ccKTXxRq.ltrans1.o:(.text.startup+0x11c): undefined reference to `u8g2_font_wqy16_t_gb2312b'

ccKTXxRq.ltrans1.o:(.text.startup+0x11e): undefined reference to `u8g2_font_wqy16_t_gb2312b'

ccKTXxRq.ltrans1.o:(.text.startup+0x132): undefined reference to `u8g2_font_wqy16_t_gb2312b'

C:\Users\Lenovo\AppData\Local\Temp\ccKTXxRq.ltrans1.ltrans.o:ccKTXxRq.ltrans1.o:(.text.startup+0x134): more undefined references to `u8g2_font_wqy16_t_gb2312b' follow

collect2.exe: error: ld returned 1 exit status

olikraus, 2018-11-01:
Hmm... in Arduino IDE I can not choose 1.5.1
How did you install 1.5.1? Installing a library is a little bit complicated: You must ensure that all code of the old library is removed.
ok, I will create 1.5.2. Maybe there is a bug with the Arduino IDE library updated.

If you want to update the lib manually: Please use this .h file and replace all other versions of the same .h file on your hard disk:
https://github.com/olikraus/U8g2_for_Adafruit_GFX/blob/master/src/u8g2_fonts.h