olikraus/U8g2_for_Adafruit_GFX #13: drawUTF8 but one character at a time

FrenchLab47, 2018-11-08:
Hello,

I use your superb library to display text in French on a 64x32 LED matrix
For your information, the operation is perfect with the other libraries for matrix of LED like:
ESP32RGBmatrixPanel
And ESP32-RGB64x32MatrixPanel-I2S-DMA which is more efficient.

With the function drawUTF8, I manage to scroll UTF8 text,,
but i would also like to display only one character at a time (animations, typewiter, slide, etc).

I tried this way with drawGlyph, but getUTF8Width value is too big.

`String tmpText = newText;   			// Variable string
tmpText.toCharArray(buf, 512);  		
utf8Ascii(buf);  						// Convert UTF8 To ascii
strcpy(disp_text, buf);
int charWidth = 7; 					// Width in pixel for one char

int8_t nbChar = matrix_u8g2.getUTF8Width(disp_text);
    
for (int i = 0; i < nbChar; i++) {
	matrix_u8g2.drawGlyph((i * charWidth), yPos, disp_text[i]);	
	/*
	 	With drawGlyph the characters are well displayed one by one,
		but nbChar is bigger because of the encoding, and there is still too much space . 
	*/	
}`

Thank you so much.

olikraus, 2018-11-08:
Just to clarify this: I think, what you want to know is the delta-x distance, which is the number of pixel to go to the right, where the next char has to be placed. This is different from the width of the gylph. For example the blank char has a width of zero, but still has a none-zero delta-x distance. Same is true for the ".", which usually has a much larger delta-x value to separate sentences. 

The good news is this: drawGlyph will return the delta-x value. So your loop can be written as this:
```
x = 0;
for (int i = 0; i < nbChar; i++) {
  x = x + matrix_u8g2.drawGlyph(x, yPos, disp_text[i]); }
```

FrenchLab47, 2018-11-08:
it is exactly that, and with a simple solution.
Big thanks :)

closed 2019-06-20 by olikraus