olikraus/U8g2_for_Adafruit_GFX #18: st7735 support

hznupeter, 2019-08-05:
How to support ST7735? can you give small examples ,thank you

hznupeter, 2019-08-07:
```
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library for ST7735
#include <SPI.h>
#include <U8g2_for_Adafruit_GFX.h>

//#define TFT_MOSI 23(sda)
//#define TFT_SCLK 18(scl)
#define TFT_CS   P16  // Chip select control pin
#define TFT_DC   P15   // Data Command control pin
#define TFT_RST  P13  // Reset pin (could connect to RST pin)
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_RST);
U8G2_FOR_ADAFRUIT_GFX u8g2_for_adafruit_gfx;
void setup(void) {
  tft.initR(INITR_BLACKTAB);      // Init ST7735S chip, black tab
  tft.setRotation(0);
  tft.fillScreen(ST77XX_BLACK);
  u8g2_for_adafruit_gfx.begin(tft);
}

void loop() {
  u8g2_for_adafruit_gfx.setFontMode(0);                 // use u8g2 none transparent mode
  u8g2_for_adafruit_gfx.setFontDirection(0);            // left to right (this is default)
  u8g2_for_adafruit_gfx.setForegroundColor(ST77XX_YELLOW);      // apply Adafruit GFX color
  u8g2_for_adafruit_gfx.setFont(u8g2_font_wqy14_t_gb2312 );  // select u8g2 font from here: https://github.com/olikraus/u8g2/wiki/fntlistall
  u8g2_for_adafruit_gfx.setCursor(20, 20);               // start writing at this position
  u8g2_for_adafruit_gfx.print("世界，你好");
  delay(1000);
}

```

closed 2019-08-07 by hznupeter

olikraus, 2019-08-07:
Looks good.