olikraus/U8g2_for_Adafruit_GFX #19: there is no SetFontPosTop function in the library

hznupeter, 2019-08-20:
there are servral function to set the font postition in the u8g2 library.But it can't use the the U8g2_for_Adafruit_GFX.
here is the lost function:
`void setFontPosBaseline(void) { u8g2_SetFontPosBaseline(&u8g2); }
void setFontPosBottom(void) { u8g2_SetFontPosBottom(&u8g2); }
void setFontPosTop(void) { u8g2_SetFontPosTop(&u8g2); }
void setFontPosCenter(void) { u8g2_SetFontPosCenter(&u8g2); }`
can i add these functions into this library, thk bro.

olikraus, 2019-08-20:
Indeed I simplified this version.

> can i add these functions into this library, thk bro.

sure...

hznupeter, 2019-08-21:
the fontpos is very useful .(*￣︶￣)

hznupeter, 2019-09-05:
can you add `setFontPosTop` function in this repo. I've tried a lot,but failed. thank you

olikraus, 2019-09-07:
You just need to add `getFontAscent()` to the y value to get the baseline position from the top pos position. 

The following function  will accept the upper left position of instead of the base line position. In the same way you can write wrapper functions to other functions if required.

```
int16_t drawTopPosUTF8(U8G2_FOR_ADAFRUIT_GFX &u8g2, int16_t x, int16_t y, const char *str);
{
   return u8g2.drawUTF8(x,  y+u8g2.getFontAscent(),str);
}
```

code is not tested...