olikraus/U8g2_for_Adafruit_GFX #20: Use a pointer to a display object for begin

rlightner, 2019-09-07:
I'm using a pointer to an object:
display->init();
u8g2Fonts->begin(display);

where display is:
GxEPD2_BW<GxEPD2_270, GxEPD2_270::HEIGHT> *display;
But I'm getting an error: 
no matching function for call to 'U8G2_FOR_ADAFRUIT_GFX::begin(GxEPD2_BW<GxEPD2_270, 264u>*&)'

Any thoughts? Thanks!

olikraus, 2019-09-07:
You have to pass the object instead of the pointer:
`u8g2Fonts->begin(display);`

rlightner, 2019-09-07:
I originally tried that, but got an error:
`Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.`
which of course I had no idea what that means. 

I'll keep working on this :/

ZinggJM, 2019-11-09:
@rlightner 

I think your u8g2Fonts is a NULL pointer, unless you have assigned the address of an U8G2_FOR_ADAFRUIT_GFX instance anywhere. Don't wake sleeping Guru's.

rlightner, 2019-11-09:
Sorry, this was asked back in September, you’ve been sleeping a while!

closed 2019-11-09 by rlightner

olikraus, 2019-11-09:
So you fixed this problem meanwhile?

rlightner, 2019-11-09:
I ended up refactoring the code to a specific module so I didn't pass around display.  @ZinggJM hope you didn't take my comment the wrong way! I think of you as a Guru, so when I saw the "Don't wake sleeping Guru's." comment, I thought you were referring to yourself and not the actual bug!