olikraus/U8g2_for_Adafruit_GFX #22: Add bigger Cyrillic font(s) please

ZinggJM, 2019-11-09:
Hi Oliver,

could you add bigger Cyrillic font(s) to u8g2 and U8g2_for_Adafruit_GFX?
Please see Arduino Forum Topic https://forum.arduino.cc/index.php?topic=644866.0

Thank you

Jean-Marc

olikraus, 2019-11-09:
Hmmm the only option is to derive the gylphs from a ttf font. inconsolata contains cyrillic glyphs.
Whould this be ok? I can several sizes up to 62 pixel height...

[image]

olikraus, 2019-11-09:
ok, looks like the 62 pixel height version does not work. Will create up to 49 pixel height

ZinggJM, 2019-11-09:
Thank you very much. As you see, this is not for me, and I am not a font expert. I am happy with anything you can provide.

olikraus, 2019-11-09:
Yes, I know. It's done already, but I still have to update this project. Hope to do this over the weekend.

closed 2019-11-10 by olikraus