olikraus/U8g2_for_Adafruit_GFX #26: Fonts not declared

el-samiyel, 2020-09-25:
Hi, there seems to be an issue with the new fonts available in the standard u8g2 lib.

When using the u8g2_for_Adafruit_GFX i get "***font not defined in this scope".

How can I update this lib to work with the new fonts?

Cheers

olikraus, 2020-09-25:
Well, yes, this project is a little bit behind the latest version of u8g2.
Which font did you miss in detail?

el-samiyel, 2020-11-08:
Hi Olikraus,

u8g2_font_lubR14_tf. Is there a way for me to generate these fonts?

Cheers

ftjuh, 2025-03-27:
Coming late to the rescue:

I had the same problem and with support from GPT 4o I created an updated version of `U8g2_for_Adafruit_GFX.h` which now reflects the current `u8g2` font list. This covers step 3 of the [Release Procedure](https://github.com/olikraus/U8g2_for_Adafruit_GFX?tab=readme-ov-file#release-procedure) described on the main page, which I found to be a bit unclear at first, so here's what you do:

1. From the `u8g2` repository, download the current [`u8g2_fonts.c`](https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_fonts.c) (big file, it contains all the font bitmaps).
2. Edit it and change the included file name from "`u8g2.h`" to "`u8g2_fonts.h`".
3. Save it to the /src folder of this library, overwriting the old version.
4. Copy the attached 'U8g2_for_Adafruit_GFX.h' in the same folder, also overwriting the old version.

I'll gladly post this as a commit, if it helps. Thanks, Oli, for all the effort you put into this.

[U8g2_for_Adafruit_GFX.zip](https://github.com/user-attachments/files/19492801/U8g2_for_Adafruit_GFX.zip)

.

olikraus, 2025-04-11:
Thanks for posting this. Truly, I should update this repo...

tookitogo, 2026-10-08:
> Thanks for posting this. Truly, I should update this repo...

Yes, _please!!_