olikraus/U8g2_for_Adafruit_GFX #29: Hi, I have some 40KB fonts. How can I save them in flash of esp8266? I've tried to add "PROGMEM" definition, but I will report an error

jie326513988, 2021-02-12:
Thank you very much for your library！

esp8266 12-f
arduino ide version 1.8.13

Or how to use the file system to call the font? The font file can be saved in the file system, but I don't know how to call it

[image]
[image]

jie326513988, 2021-02-12:
I have solved it myself, just copy U8G2's memory usage definition function for ESP8266

ppelikan, 2021-05-21:
I have provided solution in the following pull request: https://github.com/olikraus/U8g2_for_Adafruit_GFX/pull/21

olikraus, 2021-05-24:
> I have provided solution in the following pull request: #21

Indeed, but you didn't answer my questions there. Can't we takeover the u8g2 code?

ppelikan, 2021-05-24:
I don't see any questions from you there. There are no comments at all.
My fix has been tested and works ok, but if you prefer another solution, then you can close my pull request :)

jayxtt999, 2021-09-04:
> 我自己解决了，直接复制U8G2的ESP8266的内存使用定义函数

是B站的甘草大佬吗。害。。。我都想换falsh了

jie326513988, 2021-09-04:
> > 我自己解决了，直接复制U8G2的ESP8266的内存使用定义函数
> 
> 是B站的甘草大佬吗。害。。。我都想换falsh了

也可以直接用我改好的库哦

jayxtt999, 2021-09-04:
> > > 我自己解决了，直接复制U8G2的ESP8266的内存使用定义函数
> > 
> > 
> > 是B站的甘草大佬吗。害。。。我都想换falsh了
> 
> 也可以直接用我改好的库哦

似乎gitee里面也没有改过的库，群也满员了。。。

jayxtt999, 2021-09-04:
> > > 我自己解决了，直接复制U8G2的ESP8266的内存使用定义函数
> > 
> > 
> > 是B站的甘草大佬吗。害。。。我都想换falsh了
> 
> 也可以直接用我改好的库哦

大佬牛逼,躺了。

jie326513988, 2021-12-26:
> 我自己解决了，直接复制U8G2的ESP8266的内存使用定义函数

需要添加的内容为
https://gitee.com/Lichengjiez/weather-ink-screen/raw/master/%E5%9B%BE%E7%89%87/GFXFont1.png
https://gitee.com/Lichengjiez/weather-ink-screen/raw/master/%E5%9B%BE%E7%89%87/GFXFont2.png

wei4118268, 2024-01-03:
It works! 牛了大兄弟。你给的gitee的图片链接打开是错的呢，我还是去gitee自己找的哈哈。