olikraus/U8g2_for_Adafruit_GFX #31: Error compiling when included alongside U8g2lib.h

Spirik, 2021-07-25:
Hello, @olikraus!

I'm having a rather edge case issue, when trying to compile a sketch, libraries in which can be simultaneously dependent on both `U8g2lib` and `U8g2_for_Adafruit_GFX`: I get a lot of `multiple definition` errors.

[image]

To reproduce, just include both libraries in your sketch:
```
#include <U8g2lib.h>
#include <U8g2_for_Adafruit_GFX.h>
```

I am writing [GEM](https://github.com/Spirik/GEM) library for managing graphic multi-level menus. It currently supports `AltSerialGraphicLCD` and `U8g2` libraries. And I am in a process of adding support for `Adafruit GFX` library. And I'd like to implement fonts via `U8g2_for_Adafruit_GFX`, so that visually (and partially code-wise and feature-wise) menu look similar to what user would get from `U8g2` version.

So I include all of the above libraries in corresponding .h files of the GEM (`GEM.h`, `GEM_u8g2.h`, `GEM_adafruit_gfx.h`). While user is expected to include only one of them in their sketch, at a compilation step compiler would require all of the libraries above (unless they explicitly excluded from [`config.h`](https://github.com/Spirik/GEM#configuration) file). It won't affect compiled sketch size (as only needed version of library will be linked), however the compiler will throw the above error. 

And again, I understand, that this is very narrow issue and unlikely to be relevant for many other users. So my plan B solution to the problem will be to just stick with the default font engine that `Adafruit GFX` library provides.

Either way, thank you for the great libraries you've created 🙏

olikraus, 2021-07-25:
You are right, u8g2 and this lib will not work together 😟

Spirik, 2021-07-25:
Ok, understood!

The followup (and slightly offtopic) question: can you suggest any possible way to convert some of the U8g2 fonts to Adafruit GFX format? I am using [`u8g2_font_6x12_tr`](https://raw.githubusercontent.com/wiki/olikraus/u8g2/fntpic/u8g2_font_6x12_tr.png) and [`u8g2_font_tom_thumb_4x6_tr`](https://raw.githubusercontent.com/wiki/olikraus/u8g2/fntpic/u8g2_font_tom_thumb_4x6_tr.png) for menu items and titles. That way I at least could maintain similar visual appearance of the menu.

closed 2021-08-08 by Spirik

olikraus, 2021-08-08:
> The followup (and slightly offtopic) question: can you suggest any possible way to convert some of the U8g2 fonts to Adafruit GFX format? I am using u8g2_font_6x12_tr and u8g2_font_tom_thumb_4x6_tr for menu items and titles. That way I at least could maintain similar visual appearance of the menu.

I am not aware of any conversion method here. At least the bdf sources of the mentioned fonts are available in the u8g2 repo. Not sure if this helps.

Spirik, 2021-08-08:
@olikraus Thanks! That's pretty much exactly what I ended up doing - converting original bdf files to Adafruit GFX format via Python script (and then tweaked resulted code manually).