olikraus/U8g2_for_Adafruit_GFX #35: Missing functions

millenseer, 2022-01-26:
Really happy to display Umlauts on my ST7789 TFT display!

But I am missing some functions from the standard Adafruit GFX?
- setTextWrap()  (badly missing this one)
- fillScreen()  (OK, this one is probably not required)

What exactly are the parameters for setCursor() ? x and y using the font size? 

Thx

olikraus, 2022-01-26:
setCursor: https://github.com/olikraus/u8g2/wiki/u8g2reference#setcursor

fillScreen: You can still use the original Adafruit procedures.

text wrap: https://github.com/olikraus/u8g2/issues/1081

millenseer, 2022-01-26:
Thanks for your kind support. The missing text wrap is really unfortunate. Is there a function to get the width of a character using the specified font? Maybe I can write my own text wrap.

millenseer, 2022-01-26:
Just saw, that the AdaFruit code for print() already works with \r and \n. Hence I gave this a try:

```
String wordWrap(String s, int limit) {

  int space = 0;
  int i = 0;
  int line = 0;
  while (i < s.length()) {
    if (s.substring(i, i + 1) == " ") {
      space = i;
    }
    if (line > limit - 1) {
      s = s.substring(0, space) + "~" + s.substring(space + 1);
      line = 0;
    }
    i++; line++;
  }
  s.replace("~", "\n");
  return s;
}

u8g2_for_adafruit_gfx.print(wordWrap("This is a test with Umlauts äöüßÖÄÜ",16));
```
Works! Almost happy. The limit is hard coded to 16 but needs to be more dynamic asking the fontsize and fontwidth. Any idea?

olikraus, 2022-01-27:
https://github.com/olikraus/U8g2_for_Adafruit_GFX/blob/82d2b3eea866e7d40266672b41e5c8306ee97403/src/U8g2_for_Adafruit_GFX.h#L173

You can use getUTF8Width to get the width of a given string.

Please note that strings are usually UTF-8 encoded. This means that for example  German umlaut chars are encoded with two bytes (see here: https://www.compart.com/de/unicode/U+00E4). 
This also means that normal string processing procedures may not work as expected. For example s.length() will not return the number of visible characters (instead it returns the number of bytes in the string).
Same is true for "~" which actually is a sequence of three bytes. So the replacement is very time consuming.

millenseer, 2022-01-27:
I was hoping for a function which returns the width of a char or a string in pixels it will consume on the display. But then again, I am happy with my wordWrap. No need to calculate the pixel width, which would probably only make sense when using fonts with flexible widths. (TTF?)
The fillScreen() to clean the display takes more time than the string replacement :-) I was looking for a cheap marker which is very uncommon in written text, thought that "~" might be one.

olikraus, 2022-01-27:
> I was hoping for a function which returns the width of a char or a string in pixels it will consume on the display. 

This is exactly what "getUTF8Width" will do, it returns the number of pixel, which will be required by the provided UTF8 string on the display.

> which would probably only make sense when using fonts with flexible widths. (TTF?)

Almost all fonts in u8g2 have variable character width. Monospaced fonts in u8g2 are marked with "m" at the end.

> that "~" might be one.

Sure, but `s.replace("~", "\n");` has to replaced three bytes by one byte, so a lot of array copying is required.

CarlosFly74, 2025-02-22:
Hi! Does this directory not have setPowerSave?

olikraus, 2025-02-22:
sertPowerSave is a display specific function. This project just addes some fonts...

closed 2026-11-08 by marsrakete