olikraus/U8g2_for_Adafruit_GFX #36: Error compiling Platform IO

jerome6994, 2022-02-07:
Hello

Thank you for this work which should normally help me to set up the degree symbol °
But I get the following error when compiling:
redefinition of 'struct _u8g2_font_info_t'
redefinition of 'struct _u8g2_font_decode_t'

I'm on an ESP8266 Lolin D1 mini board with a TCA9548A board and 3 OLEDs behind (2 x SSD1306 and 1 SH1106)

Thanks for your help

olikraus, 2022-02-07:
"redefinition"? sound more like a local problem. Maybe this lib is included twice.

jerome6994, 2022-02-07:
Thank you very much for your quick feedback
So it is true that I have these libraries there because I am testing a solution not to redo all my code to allow me to display the symbol ° degree
[image]

jerome6994, 2022-02-07:
by replacing the line 
`#include "U8g2lib.h"`
by
`//#include "U8g2lib.h"`
I don't have the error anymore.

thanks for the info for a newbie like me

Good day to you

jerome6994, 2022-02-07:
I also have a strange phenomenon again!
When I call the police with for example:
`u8g2.setFont(u8g2_font_helvR14_tf);`
or as soon as I want to write on the screen
`u8g2.print(F("INT TEMP"));`

this makes me reboot my webserver and of course it does not display anything behind
I don't really understand why because normally no particular link between the two except for the HTTP request that I send and that I want to write on the screen

then another element I go through a TCA9548A multiplexer card

olikraus, 2022-02-07:
> by replacing the line `#include "U8g2lib.h"` by `//#include "U8g2lib.h"` I don't have the error anymore.

Yes, correct, you can not use u8g2lib and U8g2_for_Adafruit_GFX in parallel.

> u8g2.print(F("INT TEMP"));

F() and .print() are functions from the Arduino library.
Will `u8g2.print("INT TEMP");` work?

jerome6994, 2022-02-07:
I just made the change but it's the same.
I check if it is not upstream of the code in the declarations in particular of the TCA9548A card
I'm just trying to replace this code

`		else if(serverWeb.argName(0) == "msg2") {
			// j'écris sur l'OLED présent sur le bus I2C n°2
			TCA9548A(2);
			display2.clearDisplay();			
			display2.setTextSize(2);
			display2.setTextColor(WHITE);
			// Display static text
			Display21Center("TEMP INT");
			display2.display();
			// j'écris la partie message de la requète sur l'écran
			display2.setTextSize(3);
			Display22Center(message2);
			display2.drawCircle(80, 30, 2, WHITE);
			display2.setCursor(85, 30);
			display2.print("C");
			display2.setCursor(0, 50);
			display2.display();
			Serial.println(message2);
			// On repond au client
			serverWeb.send(200, "text/plain", "Message reçu : " + message2);	
		}`

With this one

`		else if(serverWeb.argName(0) == "msg2") {
			// j'écris sur l'OLED présent sur le bus I2C n°2
			TCA9548A(2);
			display2.clearDisplay();			
			u8g2.setFontMode(1);                 // use u8g2 transparent mode (this is default)
			u8g2.setFontDirection(0);            // left to right (this is default)
			u8g2.setForegroundColor(WHITE);      // apply Adafruit GFX color
			u8g2.setFont(u8g2_font_helvR14_tf);  // select u8g2 font from here: https://github.com/olikraus/u8g2/wiki/fntlistall
			u8g2.setCursor(0,20);                // start writing at this position
			u8g2.print("TEMP INT");
			u8g2.setFont(u8g2_font_helvR24_tf);  // select u8g2 font from here: https://github.com/olikraus/u8g2/wiki/fntlistall
			u8g2.setCursor(0,60);                // start writing at this position
			u8g2.print(message2 + "°C");            // UTF-8 string with german umlaut chars
			display2.display();                                   // make everything visible                      // make everything visible
			Serial.println(message2);
			// On repond au client
			serverWeb.send(200, "text/plain", "Message reçu : " + message2);	
		}`

olikraus, 2022-02-07:
Not clear how I can support here.

jerome6994, 2022-02-08:
Hello
yes I am aware of it already it was to know if my replacement was correct.
then to know you would need the entire code, right?
are you ok with that?
To see a beginner's code without falling off your chair because there must be a lot of error

olikraus, 2022-02-08:
Unfortunately my knowledge on Adafruit GFX, webservers and your requirements is almost zero.
So any kind of review would be extremly time consuming. You can always ask specific question about u8g2, but the effort on doing a review on a user code is too high.