olikraus/U8g2_for_Adafruit_GFX #37: Facing problems in displaying a struct-initialized char array on the e-ink display

faisal-saddique, 2022-09-11:
Hello olikraus, I'm trying to do something like this:

`struct myStruct { // creating a struct to store the individual words of the label
  char number[6];
  char id[6];
  char type[6];
  char description[14];
  char sig[6];
  char date[18];
};

struct myStruct data;
data.description="random";

U8G2_FOR_ADAFRUIT_GFX u8g2Fonts;

u8g2Fonts.print(data.description);`

The code may not compile as this is just a concept of what I am trying to do over here. The problem I am facing over here is that it doesn't print anything.

olikraus, 2022-09-11:
Your code seems to ok.

faisal-saddique, 2022-09-11:
Yes the issue is resolved. I had a bug in my function. Thanks for replying though.

closed 2022-09-11 by faisal-saddique