olikraus/U8g2_for_Adafruit_GFX #38: User exception while using multiple setFont

m42u, 2022-12-16:
Hi guys!

I'm trying to use this project to display temperature and humidity value on a Waveshare ePaper display using a DHT22. 
I'm able to display some text using the `print` function after having chosen the font using `setFont`.
Basically, this is working fine:
```
#include <GxEPD2_BW.h>
#include <U8g2_for_Adafruit_GFX.h>

GxEPD2_BW<GxEPD2_290_T94, GxEPD2_290_T94::HEIGHT> display(GxEPD2_290_T94(/*CS=15*/ SS, /*DC=4*/ 4, /*RST=2*/ 2, /*BUSY=5*/ 5));

U8G2_FOR_ADAFRUIT_GFX u8g2Fonts;

void setup() {
  display.init();
  u8g2Fonts.begin(display);
  display.setRotation(1);
  u8g2Fonts.setFontMode(1);
  u8g2Fonts.setFontDirection(0);
  u8g2Fonts.setForegroundColor(GxEPD_BLACK);
  u8g2Fonts.setBackgroundColor(GxEPD_WHITE);

  display.setFullWindow();
  display.firstPage();
  do
  {
    display.fillScreen(GxEPD_WHITE);
    u8g2Fonts.setFont(u8g2_font_logisoso58_tf);
    u8g2Fonts.setFontMode(1);
    u8g2Fonts.setCursor(0, 0);
    u8g2Fonts.print("test");
  }
  while (display.nextPage());
}
```

However, if I change the font using the `setFont` function, I got a "User exception (panic/abort/assert)".
```
[...]
    u8g2Fonts.setCursor(0, 0);
    u8g2Fonts.print("test");
    u8g2Fonts.setFont(u8g2_font_logisoso26_tf); # Here
  }
  while (display.nextPage());
[...]
```

Stack trace:
```
17:30:20.393 -> Connecting to MQTT... 
17:30:20.535 -> User exception (panic/abort/assert)
17:30:20.535 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
17:30:20.535 -> 
17:30:20.535 -> Unhandled C++ exception: OOM
17:30:20.535 -> 
17:30:20.535 -> >>>stack>>>
17:30:20.535 -> 
17:30:20.535 -> ctx: cont
17:30:20.570 -> sp: 3ffffdc0 end: 3fffffc0 offset: 0000
17:30:20.570 -> 3ffffdc0:  402550e0 3fff458c 3fff4276 00000000  
17:30:20.570 -> 3ffffdd0:  000000fe 00000000 00000000 00000000  
17:30:20.570 -> 3ffffde0:  00000000 00000000 00000000 00000001  
17:30:20.570 -> 3ffffdf0:  00007fff 00000000 c030c02c 00000000  
17:30:20.570 -> 3ffffe00:  00000000 4020378f 000005d8 40208dda  
17:30:20.570 -> 3ffffe10:  00000000 4020378f 00000020 40208df9  
17:30:20.607 -> 3ffffe20:  c02ac026 c00ec004 000005d8 402083dc  
17:30:20.607 -> 3ffffe30:  c09dc09c 00000001 3fff4a9c 4020378f  
17:30:20.607 -> 3ffffe40:  00000000 c00dc003 3fff000a 402033f8  
17:30:20.607 -> 3ffffe50:  00000000 00000000 00000d60 3fff26bc  
17:30:20.607 -> 3ffffe60:  00000000 3fff6c34 3fff4a9c 40203c39  
17:30:20.607 -> 3ffffe70:  0000007f 3fff3ef0 00000001 4020858c  
17:30:20.607 -> 3ffffe80:  00000000 00000000 000022b3 40208f8e  
17:30:20.607 -> 3ffffe90:  00000000 3fff695c 3fff4a9c 40202b6e  
17:30:20.607 -> 3ffffea0:  3fff6534 00000002 3ffffef4 420ccccd  
17:30:20.641 -> 3ffffeb0:  000022b3 3fff26bc 3fff4a9c 420ccccd  
17:30:20.641 -> 3ffffec0:  000022b3 3fff26bc 3fff4a9c 40203dc5  
17:30:20.641 -> 3ffffed0:  4020c974 326e3634 4020c974 326e3634  
17:30:20.641 -> 3ffffee0:  3fff2698 3fff3c8c 3fff2698 4020b06a  
17:30:20.641 -> 3ffffef0:  4020735c 3ffe87c8 3fff3c8c 40204b5c  
17:30:20.641 -> 3fffff00:  4020735c 3ffe87c8 3fff2698 40204a32  
17:30:20.641 -> 3fffff10:  3fff2698 3fff3c8c 3fff3c8c 402075e8  
17:30:20.677 -> 3fffff20:  420d3333 00000005 3fff3c8c 420ccccd  
17:30:20.677 -> 3fffff30:  3fff2698 3fff3c8c 00000003 402011fe  
17:30:20.677 -> 3fffff40:  420d3333 4195999a 3fff3c8c 40201a34  
17:30:20.677 -> 3fffff50:  4195999a 2601a8c0 420d3333 420ccccd  
17:30:20.677 -> 3fffff60:  4194cccd feefeffe feefeffe feefeffe  
17:30:20.677 -> 3fffff70:  4194cccd 00000001 60000000 4041a666  
17:30:20.677 -> 3fffff80:  40000000 4032b333 feefeffe feefeffe  
17:30:20.677 -> 3fffff90:  feefeffe feefeffe feefeffe 3fff3ddc  
17:30:20.711 -> 3fffffa0:  3fffdad0 00000000 3fff3dc8 402086a4  
17:30:20.711 -> 3fffffb0:  feefeffe feefeffe 3ffe85d8 40100cb5  
17:30:20.711 -> <<<stack<<<
17:30:20.711 -> 
17:30:20.711 -> last failed alloc call: 4020378F(1496)
17:30:20.711 -> 
17:30:20.711 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
```
The strange thing is that I got this exception before reaching this function call. It happens when connecting to a MQTTS endpoint (this happens before the `display.init()` function call). I'm using the Adafruit MQTT library. 

Is there a way to use multiple fonts without facing this issue?

Thanks!

olikraus, 2022-12-16:
Hmmm... I have no idea what could be the problem

ZinggJM, 2022-12-17:
Hi

The exception says "OOM": Out Of Memory. MQTT tries to allocate memory from heap and fails.

I do not fully understand the influence of using multiple fonts. Maybe they are not used out of flash (PROGMEM).
ESP32 has enough RAM for full buffered display with GxEPD2 for the smaller e-paper displays. But it may not leave enough for MTQQ.
You can reduce memory use of GxEPD2 by reducing the page height, e.g. with GxEPD2_290_T94::HEIGHT / 2.

m42u, 2023-01-02:
Hi @ZinggJM 

Thanks, I'm now able to use a second font... but not a third one 😞 
I tried `/ 4`, `/ 6` and even `/ 8` but no luck. Any other ideas?
By the way, what's the impact of reducing this number?

Thanks!