olikraus/U8g2_for_Adafruit_GFX #39: Can we have more hebrew fonts?

Crapy, 2023-04-11:
Preferly ones with bigger sizes

olikraus, 2023-04-11:
I have not created the fonts. I just added what I found in internet. So actually I do depend on the users, telling me what fonts are required. However one problem is, that u8g2 is a monochrome library which usually requires hand crafted bitmap fonts. This means providing a true type font will not provide good results. 

Anyhow, if you have a font in mind, then it could be added.

olikraus, 2023-04-12:
> If you can add more fonts like cu12, x11, or unifont with bigger sizes (16, 18, 24) that will be very helpful.

 cu12 an unifont are hand crafted pixel fonts, they do not exist in any other size.
x11 fonts exists in multiple sizes, but do not contain hebrew glyphs.

I can't provided and add to u8g2 what doesn't exist. If can provide a suitable bdf font with the desired size and glyphs, I will be happy to add those fonts.