olikraus/U8g2_for_Adafruit_GFX #42: HUB75-MatrixPanel-DMA NOT SUPPORTED ?

bangguong84, 2023-06-05:
CAN HELP ME TO WORK WITH THE LIBRARY
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA

olikraus, 2023-06-05:
Just use it...

bangguong84, 2023-06-06:
> Just use it...

can you help me 1 example writing word "hello" using https://github.com/olikraus/U8g2_for_Adafruit_GFX via https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA
Is that Okay? because i did it but it didn't work.

bangguong84, 2023-06-06:
> In the example, replace the constructor with the constructor for your matrix panel:
> 
> https://github.com/olikraus/U8g2_for_Adafruit_GFX/blob/82d2b3eea866e7d40266672b41e5c8306ee97403/examples/Hello_Adafruit_SSD1306/Hello_Adafruit_SSD1306.ino#L18

Thank you for the quick response, I did it!