olikraus/U8g2_for_Adafruit_GFX #44: (Feature request) Draw at negative position

v-ivanyshyn, 2023-10-13:
For example, I want to achieve scroll effect for some text (so the text line scrolls from it's start to the end toward to the left). It would be easy done by offsetting text position by subtracting scroll factor: `u8g2.drawStr(strX - scrollOffset, strY, str)`. But the problem is that all position values are unsigned: `u8g2_uint_t drawStr(u8g2_uint_t x, u8g2_uint_t y, const char *s)`

**Is there any other way how to place a text or any other object at negative position while drawing?**

olikraus, 2023-10-22:
The "unsigned" data type doesn't prevent you from do the scrolling. Just go ahead and do some testing. It will work. 
You  might need to enable 16 bit version of u8g2 however (which is I think the default for the latest releases)

closed 2023-10-23 by v-ivanyshyn

olikraus, 2023-10-23:
Update: There is no need to activate 16 bit mode in this lib. 16 bit is there by default. So: No todo. I just wanted to confirm this.