olikraus/U8g2_for_Adafruit_GFX #47: Space chars do not wipe out previous content

wowa-2017, 2024-04-23:
With the monospace font, I used to redraw the number during edit, just printing the right-justified string representing the edited number. Now, I see that in U8g2_for_Adafruit_GFX leading spaces don't wipe out previously drawn digits, no matter what the transparency setting is. Seems that space chars aren't drawn, instead, the engine just advances the x position. The font is the u8g2_font_7x13_t_cyrillic, so only transparent version is available, as far as I can see.

Is that done by design? That behaviour is rather strange, because despite its transparent mode, regular chars wipe out the beneath canvas, they don't form a mix of both old/new char's pixels in one place. Thus, the workaround is to clear the rectangle beneath the string before drawing, but it's needed only if the string contains spaces, which is not much convenient. Shouldn't spaces wipe background out as the other chars do?

olikraus, 2024-04-27:
Using the font mode is not enough, you also need to select the correct font. There are special monochrome fonts, which need to be used in such a case: The second last char in the font must be "m"

wowa-2017, 2024-04-27:
It's what I wrote - 1) m-font isn't available, and 2) other chars of this t-font wipe out the background perfectly, why this exception with the space char?

olikraus, 2024-04-28:
The m font contains the bitmap for the complete glyph including the transparent part. The t fonts reduces the transparent part to the visible part of the glyph. 
In other words for a 5x7 "m" font each glyph is a bitmap with exactly 5x7 pixel. However for a "t" font the bitmap is reduced to the visible pixel area, so for example a sans-serif "i" might be reduced to a 1x6 bitmap.
Moreover for the "space" the bitmap is fully removed in a "t" font. This will reduce the font size a lot, which is important for embedded controllers with limited flash memory.

I general I would suggest a box drawing procedure to erase previously visible string. 
But you could also generate the "m" font if required. Maybe you can let me know the precise font name.

wowa-2017, 2024-04-28:
Thank you, now it's much more clear! As I wrote in the first post, I'm using your u8g2_font_7x13_t_cyrillic font. Can you make an 'm' version of it?

olikraus, 2024-05-10:
cmd: ./bdfconv -v -b 2 -f 1 -m '32-128,$400-$52f' -n u8g2_font_7x13_m_cyrillic -o u8g2_font_7x13_m_cyrillic.c ../bdf/7x13.bdf

``` C++
/*
  Fontname: -Misc-Fixed-Medium-R-Normal--13-120-75-75-C-70-ISO10646-1
  Copyright: Public domain font.  Share and enjoy.
  Glyphs: 333/3226
  BBX Build Mode: 2
*/
const uint8_t u8g2_font_7x13_m_cyrillic[5623] U8G2_FONT_SECTION("u8g2_font_7x13_m_cyrillic") = 
  "M\2\3\2\3\4\1\2\4\7\15\0\376\11\376\11\0\1\272\3\255\5W \6\357\374\371\177!\12\357"
  "\374iq\337\341\234\6\42\14\357\374II\230\204I\316\237\0#\21\357\374\231\222\60\311\6-\311\6-"
  "\11\223\234\11$\17\357\374\331\322!J\322\65\211\206\64g\3%\17\357\374)YRj\215\323RK-"
  "'\1&\20\357\374\271\210Q\26\205b\224\324\262%\247\0'\11\357\374iq\235\277\1(\15\357\374\211"
  "i\234\306\365\70\317I\0)\15\357\374Iy\234\307\325\70\315\211\0*\15\357\374\271D\241\66lb\224"
  "s\2+\14\357\374\271\305\341 \306\71\67\0,\12\357\374\371\247Q\314Q\0-\11\357\374\371\62\350|"
  "\1.\12\357\374\371\267t\315\61\0/\15\357\374\251q\32\247\345\64\316\251\0\60\21\357\374IbTL"
  "\302$L\302$\214\212:\15\61\14\357\374i\251\230\304=\16:\5\62\20\357\374)C\24&a\234\206"
  "b\32\17;\5\63\17\357\374\361aN\253{\234\204\321\220\223\0\64\21\357\374\211\251\230dQ-\312\242"
  "a\215s\22\0\65\20\357\374\361a\211+\223\26W\302h\310I\0\66\21\357\374I[\32W&-\11"
  "\223\60\32r\22\0\67\15\357\374\361aN\313i\234\306\71\25\70\24\357\374)C\24&a\22FC\24"
  "&a\22FCN\2\71\21\357\374)C\24&a\222IK\134\315v\32\0:\14\357\374\271\245kN"
  "N\327\34\3;\14\357\374\271\245kN\35\305\34\5<\11\357\374\251\275w\247\0=\13\357\374\271\17;"
  "e\330\271\0>\12\357\374)y\327\356T\0\77\17\357\374)C\24&a\234\226s\70\247\1@\24\357"
  "\374)C\24&a\22-\245\244\242DI}\310I\0A\23\357\374IbTL\302$L\206%L\302"
  "$\314)\0B\23\357\374\361A+eQ\26\15Y)\213\262d\320I\0C\16\357\374)C\24&q"
  "\217\321\220\223\0D\23\357\374\361A+eQ\26eQ\26eQ\226\14:\11E\16\357\374\361a\211\313"
  "C\26\227\207\235\2F\15\357\374\361a\211\313C\26\267\223\1G\21\357\374)C\24&q\323\22&\231"
  "\264\344\24\0H\24\357\374\361\60\11\223\60\11\223a\11\223\60\11\223\60\247\0I\14\357\374)\203\30\367"
  "q\320)\0J\14\357\374ik\334S\226\355\64\0K\22\357\374\361\60\311\242Z\22\252I\30e\245\60"
  "\247\0L\12\357\374\361\270\237\207\235\2M\24\357\374\361\60\221\24II\224\212RL\302$L\302\234\2"
  "N\24\357\374\361\60\11\23-)%QRS\302$L\302\234\2O\24\357\374)C\24&a\22&a"
  "\22&a\22FCN\2P\20\357\374\361A\12\223\60\11\223A\212\333\311\0Q\26\357\374)C\24&"
  "a\22&a\22&\245$J\242!\317\21\0R\23\357\374\361A\12\223\60\11\223AJ\302(+\205\71"
  "\5S\21\357\374)C\24&q>\344q\22FCN\2T\12\357\374)\203\30\367;\15U\24\357\374"
  "\361\60\11\223\60\11\223\60\11\223\60\11\243!'\1V\21\357\374\361\60\11\223\60\252EY\24\252v\32"
  "\0W\25\357\374\361\60\11\223\60\11\223D\251(\211\244HJ\230S\0X\22\357\374\361\60\11\243Z\24"
  "\212Q\26\25\223\60\247\0Y\16\357\374)Y\224U\302$\215\273\323\0Z\15\357\374\361aN\253j\363"
  "\260S\0[\14\357\374\221!\213\373\363\220C\0\134\15\357\374)q\36\347\345<\316)\0]\13\357\374"
  "\221!\356\337\206\34\2^\13\357\374ii\222\325\371\13\0_\11\357\374\371\357\303\216\0`\11\357\374\241"
  "<\347\277\1a\17\357\374y\31\362hP\302$\223\226\234\2b\21\357\374\361\270e\322\222\60\11\23-"
  "Iv\22\0c\16\357\374y\31\242\60\211\213\321\220\223\0d\21\357\374\251qi\251)a\22&\231\264"
  "\344\24\0e\17\357\374y\31\242\60\31\226\70\214\206\234\4f\15\357\374I[)N\207\60n\247\2g"
  "\20\357\374yYjQ\226m\371\20\205\321\20\1h\21\357\374\361\270e\322\222\60\11\223\60\11s\12\0"
  "i\14\357\374\331rT\356\70\350\24\0j\15\357\374YsT\356\224EY\66\1k\17\357\374\361\270-"
  "\252\215QV\12s\12\0l\12\357\374Ir\77\16:\5m\21\357\374yQ\262\244\224\224\222RR\312"
  "r\12\0n\20\357\374\271'\223\226\204I\230\204I\230S\0o\20\357\374y\31\242\60\11\223\60\11\243"
  "!'\1p\17\357\374\271'\223\226\204\211\226$S\134\6q\16\357\374yYjJ\230d\322\22\267\0"
  "r\13\357\374\271'[)n\247\2s\17\357\374y\31\242\60\322\245\60\32r\22\0t\15\357\374Y\342"
  "t\10\343\266l'\1u\20\357\374\271\207I\230\204I\230d\322\222S\0v\17\357\374y\311\242,\312"
  "*a\222\346\64\0w\20\357\374y\311\242,JJI)\251%\71\11x\15\357\374\271\207QQ\25\243"
  "bN\1y\21\357\374\271\207I\230\204I&-q\22FC\4z\13\357\374\271\17k\327a\247\0{"
  "\15\357\374\261-\256\207z\32\327w\4|\11\357\374iq\177\247\1}\15\357\374\221=\256\346b\36\327"
  "v\14~\12\357\374I-\245:\177\2\0\0\0\10\4d\6\367\377\377\4\0\21\357\374\241<\307\206%"
  "\216\207,\216\207\235\2\4\1\23\357\374\221(\213rhX\342x\310\342x\330)\0\4\2\22\357\374\361"
  "A\214\313c\224EY\224Eq(\1\4\3\15\357\374\322\34\34\226\270\357d\0\4\4\20\357\374I["
  "%\216\207,\316\263l'\1\4\5\22\357\374)C\24&q>\344q\22FCN\2\4\6\15\357\374"
  ")\203\30\367q\320)\0\4\7\20\357\374\221,\312rh\20\343\216\203N\1\4\10\15\357\374ik\334"
  "S\226\355\64\0\4\11\24\357\374)c\22&a\22&ZRJJI\223N\2\4\12\25\357\374\361("
  "\213\262(\213\262A\252\224\222RR\322I\0\4\13\21\357\374\361A\214\313c\224EY\224E\71\5\4"
  "\14\23\357\374\322\260\222E\265$T\223\60\312JaN\1\4\15\26\357\274\362,L\302$S\242\244)"
  "\321\222\60\11\223\60\247\0\4\16\24\357|\242P\307\302$L\302h\220+a\64\344$\0\4\17\26\357"
  "\374\361,\312\242,\312\242,\312\242,\312\242A\214C\0\4\20\24\357\374IbTL\302$L\206%"
  "L\302$\314)\0\4\21\22\357\374\361A\212\313\203\24&a\22&\203N\2\4\22\25\357\374\361A\12"
  "\223\60\11\223A\12\223\60\11\223A'\1\4\23\13\357\374\361a\211\373\235\14\4\24\26\357\374Ic\22"
  "&a\222EY\224EY\24\15K\230#\0\4\25\17\357\374\361a\211\313C\26\227\207\235\2\4\26\25"
[... cut ...]
  "\357\24\357\374Y\206\34\12\223\60\11\223LZ\342$\214\206\10\4\360\24\357|\242,\312\241\60\11\223\60"
  "\32\344J\30\15\71\11\4\361\26\357\374)Q\26\345P\230\204I\230d\322\22'a\64D\0\4\362\23"
  "\357\274\272Ca\22&a\64\310\225\60\32r\22\0\4\363\24\357\374I\335\241\60\11\223\60\311\244%N"
  "\302h\210\0\4\364\22\357|\242,\312\241\60\11\223\60\32\344\356\24\0\4\365\22\357\374)Q\26\345P"
  "\230\204I\30\15r\235\2\4\370\30\357\274\222\60\311\241\60\11\223\60\231\222()%\245$\231r\12\0"
  "\4\371\25\357\374II\230\344P\230\204\311\224DI)I\246\234\2\0";

```

wowa-2017, 2024-05-11:
Thank you very much, will try it!