olikraus/U8g2_for_Adafruit_GFX #49: 0.42 inch OLED on ESP8266 01

tomboul26, 2024-12-13:
Hello,

I try your example Hello World for SSD1306 but  for an ESP8266-01, but its display incoherent things

i didn't see where you specify the number of pixel for width and hight, for an 0.42 inch , its 72*40

The pin i use are GPIO0 for SDA and GPIO2 for SDL

My code 

`
#include "ESP8266WiFi.h"
#include <Wire.h>
#include <Adafruit_SSD1306.h>
#include <U8g2_for_Adafruit_GFX.h>

#define SCREEN_WIDTH 72 // OLED display width, in pixels
#define SCREEN_HEIGHT 40 // OLED display height, in pixels

// Declaration for an SSD1306 display connected to I2C (SDA, SCL pins)
#define SCREEN_ADDRESS 0x3C 
#define OLED_RESET -1 // Reset pin # (or -1 if sharing Arduino reset pin)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
U8G2_FOR_ADAFRUIT_GFX u8g;

void setup() {
  Serial.begin(112500);
  Wire.begin(0, 2);                  // sda, scl
  delay(2000);  

  Serial.println("OLED FeatherWing test");
   
  // SSD1306_SWITCHCAPVCC = generate display voltage from 3.3V internally
  if(!display.begin(SSD1306_SWITCHCAPVCC, SCREEN_ADDRESS)) {
    Serial.println(F("SSD1306 allocation failed"));
    for(;;); // Don't proceed, loop forever
  }
  Serial.println(F("SSD1306 allocation ok"));

  u8g.begin(display);    
  // Clear the buffer
  resetDisplay(); 
  delay(100);
  u8g.setFont(u8g2_font_helvR14_tf);  // select u8g2 font from here: https://github.com/olikraus/u8g2/wiki/fntlistall
  u8g.setFontMode(1);                 // use u8g2 transparent mode (this is default)
  u8g.setFontDirection(0);            // left to right (this is default)
  u8g.setForegroundColor(WHITE);      // apply Adafruit GFX color
  u8g.setCursor(0,0);                // start writing at this position
  u8g.print("Hello World");
  u8g.setCursor(0,20);                // start writing at this position
  u8g.print("Umlaut ÄÖÜ");            // UTF-8 string with german umlaut chars
  delay(2000);
}

void resetDisplay(){
  display.clearDisplay();
  u8g.setCursor(0, 0);
}

void loop() {
}

void resetDisplay(){
  display.clearDisplay();
  display.setCursor(0, 0);
}

void loop() {
}

` 
Its display that :
[image]

olikraus, 2024-12-13:
looks like there is an issue with your display. Try to use some Adafruit commands first to ensure, that your display works as expected.

tomboul26, 2024-12-14:
Thank you @olikraus 
Yes i find an exemple with primitiv command, and i get something much better with my screen
here : https://github.com/projetsdiy/esp-01-WiFi-Scanner-ESP-01-OLED-I2C-Display

about "Adafruit commands", you mean they are these basic commands like these?

```
static void sendcommand(unsigned char com) {
  Wire.beginTransmission(OLED_address);     //begin transmitting
  Wire.write(0x80);                         //command mode
  Wire.write(com);
  Wire.endTransmission();                    // stop transmitting
}
```

Do you know where i can find documentation about this primitiv constructor command ?

olikraus, 2024-12-14:
First you should try to display some text without U8g2_for_Adafruit_GFX.h: https://www.instructables.com/Arduino-and-the-SSD1306-OLED-I2C-128x64-Display/
Once this works, try to add U8g2_for_Adafruit_GFX.h.

tomboul26, 2024-12-15:
thank you @olikraus I really appreciate the chance to have your answers !!

I tried your library because with adafruit, it did not work properly with the format of my mini screen 0.42 inch 72x40 pixel.

Finally i find a solution by reading this issue https://github.com/olikraus/u8g2/issues/1047

with this tutorial modified to run on an esp8266-01(s) : [ER-OLEDM0.42-1_Arduino_Tutorial.zip](https://github.com/olikraus/u8g2/files/3917804/ER-OLEDM0.42-1_Arduino_Tutorial.zip)

In this tutorial, i modify 2 line

in er_loed.h line 15
`#define OLED_RST  -1 `

and in Examples_ER-OLEDM0.42-1.ino line 37
`Wire.begin(0, 2); // SDA on GPIO0, SDL on GPIO2`