olikraus/U8g2_for_Adafruit_GFX #51: drawUTF8 and getUTF8Width Return Inconsistent Widths

XvjieGit, 2025-04-27:
Hello, I have a very strange problem when writing code that calculates pagination and line break functionality. I use getUTF8Width to calculate line breaks and pagination, and drawUTF8() to display Chinese characters, but the two functions return different widths for the same kanji missing. [Software translation, please forgive me]

Font:u8g2_font_wqy12_t_gb2312

[image]

[image]

olikraus, 2025-05-01:
A good question and indeed, the manual doesn't explain the difference. However the different should be only some pixel in most cases.

* drawUTF8: The return value should be used to calculate the position where the next char should appear.
* getUTF8Width: The return value should be used to calculate proper bounding boxes. See here: https://github.com/olikraus/u8g2/wiki/u8g2reference#getstrwidth

XvjieGit, 2025-05-02:
Let me describe my requirements: Since I need to allow users to jump to any page in a TXT file while reading, I need to pre-create an index for the TXT file by paginating all text in advance rather than during reading.

Is there a way to calculate the width of text without rendering it to the screen?

olikraus, 2025-05-02:
It should be possible with both functions.